cucapra/diospyros #3: Aligned writes into output matrix

rachitnigam, 2020-01-21:
Instead of using `vector-shuffle-set!` to set the computation on the output array, create a sketch where the write is done at locations `$i`, `$i+1`, ... and so on.

This forces the sketch to never shuffle the output.

rachitnigam, 2020-01-21:
Hm, I realize now that if such a solution exists, the program cost minimization procedure will already find it.

avanhatt, 2020-01-21:
It's feasible that a solution that does this doesn't exist without more iterations/shuffles in the input matrices, right?

closed 2020-01-21 by rachitnigam

rachitnigam, 2020-01-21:
Right, but wouldn't that also always increase the program cost and therefore not be worth it?

rachitnigam, 2020-01-21:
It does find a solution when the number of iterations = 6.

commit c653a7ca (2020-01-21): Force continuous writes to C. Results in unsat. Fixes #3.