cucapra/diospyros #10: Implement dead code elimination

rachitnigam, 2020-02-12:
We should run DCE after register allocation and shuffle truncation. We generate quite a few unnecessary registers.

avanhatt, 2020-10-26:
This would be good to do for human-readability of the kernels Vincent is generating (mostly in removing the statically specified shuffle arrays that are rendered dead).

avanhatt, 2021-05-06:
There are a few different places in the code this could slot in:

Currently, `shuffle-truncation-pass.rkt` is where these shuffle indices are created, so we could handle deduplication there. Alternatively, we could add a new `dead-code-elminiation-pass.rkt` file to do just the deduplication and dead code elimination passes there. That pass would get called from `compiler.rkt`

avanhatt, 2021-05-13:
One path forward: starting in `shuffle-truncation-pass.rkt`, build a map of final are already seen, and reference an existing shuffle instead of building a new duplicate one where possible. Then have a second pass to make sure any other "dead" shuffles are removed (likely working through the list of instructions backwards, removing any shuffle declaration where you haven't seen a use).