cucapra/diospyros #50: Demo requires pypy3 installation

agent-lee, 2020-10-23:
Repro:
> cdios demo/matrix-multiply.c
Standard C compilation successful
Writing intermediate files to: compile-out
cat compile-out/spec.rkt | pypy3 src/dios-egraphs/vec-dsl-conversion.py -w 4 -p > compile-out/spec-egg.rkt
/bin/sh: pypy3: command not found
make: *** [Makefile:52: compile-out/spec-egg.rkt] Error 127
cat: compile-out/kernel.c: No such file or directory

Fixed with just installing based on this: https://doc.pypy.org/en/latest/install.html

sampsyo, 2020-10-23:
Good point… maybe the thing could fall back to normal Python if PyPy isn't installed.

avanhatt, 2020-10-23:
Thanks! Added this to the README. You can also use just normal python by passing `PY=python3` to the make invocations, added this to the README as well. 

https://github.com/cucapra/diospyros/commit/abd6104af543bbcafa04502bc9be38103f3d0256

closed 2020-10-23 by avanhatt