cucapra/diospyros #53: Tensilica header imports missing from generated kernel.c

agent-lee, 2020-10-25:
Looks like in the generated kernel.c the Tensilica header imports that support some of these typedefs and macros are missing. Maybe we can emit all the assume headers in the generated kernel.c file by default to make it self-contained?

By itself, the kernel.c when compiled generates a bunch of errors like these on our build system:

stderr: arvr/projects/surreal/flash/diospyros/src/MatMult3x3x3x3.cpp:36:3: error: unknown type name 'valign'
  valign align_a_in;
  ^
arvr/projects/surreal/flash/diospyros/src/MatMult3x3x3x3.cpp:37:33: error: use of undeclared identifier 'xb_vecMxf32'
  align_a_in = PDX_LA_MXF32_PP((xb_vecMxf32 *)a_in);
                                ^
arvr/projects/surreal/flash/diospyros/src/MatMult3x3x3x3.cpp:37:46: error: expected expression
  align_a_in = PDX_LA_MXF32_PP((xb_vecMxf32 *)a_in);

<a bunch more here>

avanhatt, 2020-10-25:
Hm, I thought I fixed this in a recent patch, will double check. Thanks for the report!

avanhatt, 2020-10-25:
ded23d2534d3c0f70c93ef91fe38fbfde19a2cb2 should add the necessary headers (pushed Thursday, you may need to rebuild the Racket executables with `make` to see the update). 

Right now it unconditionally references a [`scalar.h` file from the private repo](https://github.com/cucapra/diospyros-private/blob/master/src/scalars.h), updating it now to only reference that if needed.

agent-lee, 2020-10-25:
Oh whoops. Ok I see it now. Thanks!

avanhatt, 2020-10-25:
No problem, let me know if that doesn't work now!

Also for testing mat-mul, our `evaluation/mat-mul` directory includes harness code to test it as well. If you want to run it directly, run make in that directory and pass `KERNEL_SRC=<path> A_ROWS=3 A_COLS=3 B_ROWS=3 B_COLS=3`. Ofc feel free to just copy out the code you need if you want to keep it in Xtensa Xplorer.

agent-lee, 2020-10-25:
Looks good. Thanks!

closed 2020-10-25 by agent-lee