cucapra/diospyros #55: Only emit `scalars.h` when necessary

avanhatt, 2020-10-26:
Codegen currently emits `#include "../../../../src/scalars.h"`, only do this when necessary (when the code uses scalar negation or sgn).

agent-lee, 2020-10-27:
Also need to make it not relative include since that breaks self-contained-ness. Probably can be solved with just copying out the scalars.h file to the compile-out.

The preference for larger codebases is to qualify the include path so that it doesn't get aliased with another scalars.h in other include path. Recommend using something like #include <diospyros/scalars.h> so that it doesn't clash with someone else's #include <scalars.h>.

avanhatt, 2020-10-27:
The better fix that I started working on here is whether I can just inline the functionality in `scalars.h` into the codegen itself. Cleaner/without the overhead of a function call, but will take a bit longer, sorry! :)

avanhatt, 2020-10-27:
If inlining ends up being too annoying, I'll definitely qualify the path as you suggested, thanks!

agent-lee, 2020-10-27:
That works too! Thanks!

avanhatt, 2020-12-03:
Fixed in 8531de42d021d7bc97eca4046d6530e198ba86e3, `scalars.h` no longer needed

closed 2020-12-03 by avanhatt