cucapra/diospyros #56: multiple functions in `cdios`

avanhatt, 2020-10-26:
Support compiling multiple functions in one module in `cdios`. For simplicity, enforce the specification restrictions on the last function (which can call into other functions).

avanhatt, 2021-01-22:
Basic float-> float helper functions should now work with 4f2e68d8c11bbe78c252ad40db76d0aeb166d19e. 

- Need to test functions with non-float types
- Helpers functions do not need to be tagged with `_in` and `_out` arguments
- Functions currently will miscompute if they have early returns
- The outer-most function (with  `_in` and `_out` tags) should be last for now

agent-lee, 2021-01-22:
I might have done something wrong but for this specification:
```
#define ROWS 3
#define COLS 3

void matrix_multiply_3x3(float a[ROWS * COLS], float b[COLS * COLS],
                         float c[ROWS * COLS]) {
  for (int i = 0; i < ROWS; i++) {
    for (int j = 0; j < COLS; j++) {
      c[j * ROWS + i] = 0;

      for (int k = 0; k < COLS; k++) {
        c[j * ROWS + i] += a[k * ROWS + i] * b[j * COLS + k];
      }
    }
  }
}

void multimatrix_multiply(float a_in[ROWS * COLS], float b_in[ROWS * COLS],
                          float c_in[ROWS * COLS], float d_out[ROWS * COLS]) {

  float ab[ROWS * COLS];
  matrix_multiply_3x3(a_in, b_in, ab);
  matrix_multiply_3x3(ab, c_in, d_out);
}
```
Tool fails with trace:
```
Standard C compilation successful
Writing intermediate files to: compile-out
init:expr-expr: contract violation
  expected: init:expr?
  given: #f
  context...:
   /home/vtlee/diospyros/src/c-meta.rkt:111:12: for-loop
   /home/vtlee/.racket/7.3/pkgs/rosette/rosette/base/form/control.rkt:30:25
   /home/vtlee/diospyros/src/c-meta.rkt:145:12: for-loop
   /home/vtlee/.racket/7.3/pkgs/rosette/rosette/base/form/control.rkt:31:25
   /home/vtlee/diospyros/src/c-meta.rkt:215:0: translate-fn-decl
   /home/vtlee/.racket/7.3/pkgs/rosette/rosette/base/form/control.rkt:31:25
   "/home/vtlee/diospyros/src/c-meta.rkt": [running body]
   temp37_0
   for-loop
   run-module-instance!125
   perform-require!78
Error: Compilation aborted. cdios return error code 1.
```

avanhatt, 2021-01-24:
Whoops, looks like the issue there is actually with the local variable without an initializer (`float ab[ROWS * COLS];`). 

I just added support for that in dd8c8f792b99eec99c752bbdcd79db708020e4cb (for local arrays of floats, for scalars we can expect users to initialize their variables I think).

agent-lee, 2021-01-25:
Basic repro looks good. I think this can be closed for now. Thanks!

closed 2021-01-25 by avanhatt