cucapra/diospyros #61: scalars.h emitted as relative path breaks self-contained build

agent-lee, 2020-10-27:
For the generated kernel, the scalars.h import uses a relative path in the file. Recommend forcing the build to copy scalars.h into compile-out/include or something similar so that the compile output is self-contained.

#include <float.h>
#include <math.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <xtensa/sim.h>
#include <xtensa/tie/xt_pdxn.h>
#include <xtensa/tie/xt_timer.h>
#include <xtensa/xt_profiling.h>
#include "../../../../src/scalars.h" // scalars.h requires relative path -> would rather have #include <scalar.h> or <diosypros/scalar.h> or something similar.

avanhatt, 2020-10-27:
Related to #55

agent-lee, 2020-10-27:
Oh yeah good point. I'll comment over there.

closed 2020-10-27 by agent-lee