cucapra/diospyros #69: Emit PDX_SAPOS_FP for final stores

agent-lee, 2020-12-08:
I think at some point we fixed this issue with the flush instruction but it appears to have resurfaced again on my end.

Repro from `diospyros/utils`:
- `eigen.py --kernel multiply`
- By default produces a 3x3x3x3 matrix multiply kernel
- Running test should throw a failure since matrices do not match

Fix for kernel is to add this at the bottom of the generated kernel:
`PDX_SAPOS_FP(align_c_out, (xb_vec4Mx8 *)c_out);`

Diffing the kernel generated by master with the kernel that was generated after commit 9e0119b only shows changes to the DRAM section. The code body appears to be the same but both cases seem to be failing on my machine now without the flush instruction.

avanhatt, 2020-12-08:
Ah, right now we conditionally flush if the last write is not aligned, we can switch it to always flush. Should be a fast change.

avanhatt, 2020-12-08:
Pushed that change with https://github.com/cucapra/diospyros/commit/05826e01fb26ea484923b89a0149d203d0f01585, let me know if that's still not working! Thanks!

avanhatt, 2020-12-08:
Ah might be a second change needed to fix the ordering, one sec

agent-lee, 2020-12-08:
The default 3x3x3x3 case looks fixed on my end. I'm planning to run a larger script later for other dimensions which might catch other corner cases later. Thanks!

avanhatt, 2020-12-09:
Whoops my concern turned out to be fine, closing for now!

closed 2020-12-09 by avanhatt