cucapra/diospyros #70: 2x1 x 1x4 matrix multiply causes ISS to abort

agent-lee, 2020-12-09:
While running coverage over different matrix multiply dimensions, the following pathological case appears to generate a kernel that causes the instruction set simulator validation to abort. 

Repro configurations:

Specification definition `specification.c` :
```
/*!

  Specification file of the target kernel to be consumed by the Diosypros tool

*/

#define A_ROWS 2
#define A_COLS 1
#define B_COLS 4

void MatMult2x1x1x4(
    float a_in[A_ROWS * A_COLS],
    float b_in[A_COLS * B_COLS],
    float c_out[A_ROWS * B_COLS]) {
  for (int i = 0; i < A_ROWS; i++) {
    for (int j = 0; j < B_COLS; j++) {
      c_out[j * A_ROWS + i] = 0;

      for (int k = 0; k < A_COLS; k++) {
        c_out[j * A_ROWS + i] += a_in[k * A_ROWS + i] * b_in[j * A_COLS + k];
      }
    }
  }
}
```

Manifest definition file `diospyros.json`:
``` {"inputs": {"a": "Eigen::Matrix<float, 2, 1>", "b": "Eigen::Matrix<float, 1, 4>"}, "outputs": {"c": "Eigen::Matrix<float, 2, 4>"}, "test": "c = a * b", "name": "MatMult2x1x1x4", "namespace": "Eigen", "specification": "specification.c", "specification_kernel": "MatMult2x1x1x4", "manifest_path": "build_2x1x1x4/spec", "build": "build_2x1x1x4", "src_dir": "build_2x1x1x4/src", "include_dir": "build_2x1x1x4/include", "bin": "build_2x1x1x4/bin", "test_dir": "build_2x1x1x4/test"} ```

Call `diospyros --manifest diospyros.json` and run test and/or benchmark code. Throws the following exception during ISS:
`*WARNING* Unhandled user exception: LoadStoreAlignmentCause (0xbf26daea)`

avanhatt, 2021-01-05:
Unable to reproduce this locally, on our machine execution succeeds:

```
[amv94@gorgonzola utils]$ python3 eigen.py --kernel multiply --input_rows 2 --input_cols 1 --output_cols 4 &> /dev/null
[amv94@gorgonzola utils]$ cd build/
[amv94@gorgonzola build]$ xt-xc++ bin/* -I include/ -I /usr/local/include/eigen-3 -I ../../evaluation/src &> /dev/null
[amv94@gorgonzola build]$ xt-run a.out 
Unoptimized Eigen cycles: 62228
Unoptimized Specification cycles: 773
Optimized Specification cycles: 203
```

However, I suspect the issue is that when we are loading data to vectors, we load 16 bytes from the first input when we should only load 8:
```
PDX_LAV_MXF32_XP(a_in_0_4, align_a_in, (xb_vecMxf32 *)a_in_mut, 16);
```

agent-lee, 2021-01-05:
Hmm ok let me try again with a newer pull of the repo.

agent-lee, 2021-01-05:
Hmm. So I reran this on mine and running the test validation appears to be okay but the benchmark still throws the warning at the end. I guess since this is a warning and not actually causing a functional error maybe we can ignore this for now. If it becomes a problem later maybe we can revisit it again later.

closed 2021-01-05 by agent-lee