cucapra/diospyros #78: Support for exponentiation in cdios

avanhatt, 2020-12-10:
We can support squares written as `n * n`, should be fairly simply to extend to target Racket's `exp`.

jasperxfliang, 2021-03-11:
A simple test case I would want cdios to be able to handle:
```
#define SIZE 8
void cube(float a_in[SIZE], float b_out[SIZE])
{
for (int i = 0; i < SIZE; i++)
  {
    b_out[i] = powf(a_in[i], 3);
  }
}
```
cdios currently gives the following:
```
powf: undefined;
 cannot reference an identifier before its definition
```
In the `translate` function, I think the `expr:call?` case will need to handle the `powf` function.

avanhatt, 2021-03-11:
Example test case looks spot on! 

To simplify things, it's fine to:
1. Require that the test case declares the relevant function (including the line `float powf(float x, float y);` before the use). This will let gcc preprocess the program without an issue, without requiring a header import.
2. Hardcode `powf` as a specific case that compiles to an exponent.

closed 2021-03-19 by avanhatt