cucapra/diospyros #90: CDIOS: Compiling C->Racket failed

canlandebigbaby, 2021-03-23:
cdios cdios-tests/matrix-multiply.c
Standard C compilation successful
Writing intermediate files to: compile-out
CDIOS: Compiling C->Racket failed
src/utils.rkt:47:25: current-oracle: unbound identifier
  in: current-oracle
  context...:
   /home/diospyros/Downloads/rosette/rosette/base/form/module.rkt:16:0

I just started to learn Racket programming. In the environment of Racket8.0, the above error occurred. I don't think it is a problem with the code itself. I just started to learn Racket programming. In the environment of Racket8.0, CDIOS has the above error. I don't think it is a problem with the code itself. I installed the necessary components according to the instructions, but some of them may be missed. Please Give me some suggestions, thank you！

avanhatt, 2021-03-23:
Hi, can you confirm whether you have Rosette installed (from the README, the prerequisite `raco pkg install rosette`). Thanks!

canlandebigbaby, 2021-03-26:
Thank you! I have installed Rosette, but it displays the same error.

avanhatt, 2021-04-30:
Sorry for the slow response! 

It turns out that the issue is that the new release of Rosette that is installed with Racket 8.0 is not backwards compatible. We have updated our code to use Racket 8.0 and new Rosette in 15f1528867172986f267190b31b27bd60c7ea2fc. Feel free to re-open if you run into other issues.

closed 2021-04-30 by avanhatt