cucapra/diospyros #113: Mistake in Egg optimization

JonathanDLTran, 2021-09-30:
The output after Egg optimization produces a sequence of Egg expressions nodes that do not correspond to any intended pattern in the conversion back to LLVM IR. 

Input:
`RecExpr { nodes: [Symbol("no-array-name7"), Symbol("1,"), Get([0, 1]), Add([2, 0]), Symbol("no-array-name8"), Symbol("2,"), Get([4, 5]), Symbol("no-temp-name6"), Mul([6, 7]), Symbol("no-array-name9"), Symbol("2,"), Get([9, 10]), Add([11, 8]), Symbol("no-array-name10"), Symbol("3,"), Get([13, 14]), Symbol("no-array-name11"), Symbol("2,"), Get([16, 17]), Mul([15, 18]), Num(0), Vec([3, 12, 19, 20])] }`

Output:
`[Symbol("no-array-name7"), Symbol("1,"), Get([0, 1]), Symbol("no-array-name9"), Symbol("2,"), Get([3, 4]), Num(0), Num(0), Vec([2, 5, 6, 7]), Num(1), Symbol("no-array-name8"), Symbol("2,"), Get([10, 11]), Symbol("no-array-name10"), Symbol("3,"), Get([13, 14]), Num(1), Vec([9, 12, 15, 16]), Symbol("no-array-name7"), Symbol("no-temp-name6"), Symbol("no-array-name11"), Symbol("2,"), Get([20, 21]), Num(0), Vec([18, 19, 22, 23]), VecMAC([8, 17, 24])]`

Error:
`thread '<unnamed>' panicked at '"no-array-name7"', src/lib.rs:398:15`
`Symbol("no-array-name7")` should always be paired with an array offset Symbol Node and a Get Node. In this situation, the `Symbol("no-array-name7")` should not exist by itself.

sampsyo, 2021-09-30:
Interesting… maybe I'm just being slow, but I guess I expected that these would be legible as S-expressions instead of this flat symbol list. (Put differently, I'm not exactly sure how to read this `RecExpr` dump as an expression.)

Is there a chance that this is the output of just formatting a `RecExpr` directly (i.e., `.to_string()`), but calling [the `pretty()` method on `RecExpr`](https://docs.rs/egg/0.6.0/egg/struct.RecExpr.html#method.pretty) would yield a more legible S-expression?

JonathanDLTran, 2021-10-02:
I should have started off by printing the S-expressions, and I realized that the mistake arises in how I convert the LLVM IR to Egg expressions, so I have changed the title of the issue to be less misleading.

The new S-expression input is: 
`(VecMAC
  (Vec
    no-temp-name4
    (Get
      no-array-name4
      1,)
    0
    0)
  (Vec
    1
    no-temp-name5
    (Get
      no-array-name5
      1,)
    1)
  (Vec
    no-temp-name4
    (Get
      no-array-name3
      1,)
    (Get
      no-array-name6
      3,)
    0))`
and the new S-expression output is
`(VecMAC
  (Vec
    (Get
      no-array-name7
      1,)
    (Get
      no-array-name9
      2,)
    0
    0)
  (Vec
    1
    (Get
      no-array-name8
      2,)
    (Get
      no-array-name10
      3,)
    1)
  (Vec
    no-array-name7
    no-temp-name6
    (Get
      no-array-name11
      2,)
    0))`

The corrected reason for the problem is that no-array-7 appears by itself as a node in the input S-expression.

sampsyo, 2021-10-03:
Huh, interesting… this might be dumb, but should I be surprised that the symbol `no-array-name7` _only_ appears in the output? It seems weird that egg optimization would "invent" a new array instead of just rearranging accesses to existing arrays…

JonathanDLTran, 2021-10-03:
This is completely my mistake; I made a copy and paste error, because the correct initial input should be:
`
(Vec
  (+
    (Get
      no-array-name7
      1,)
    no-array-name7)
  (+
    (Get
      no-array-name9
      2,)
    (*
      (Get
        no-array-name8
        2,)
      no-temp-name6))
  (*
    (Get
      no-array-name10
      3,)
    (Get
      no-array-name11
      2,))
  0)`

sampsyo, 2021-10-04:
Got it. I'm still not 100% clear on why the conclusion is that the LLVM->egg translation is wrong... (or did I misunderstand)? Can you point to the subexpression that was generated incorrectly?

closed 2022-03-11 by avanhatt