cucapra/diospyros #128: Issue with Logical XOR in C Program

jingyao-zhang, 2023-09-21:
Hello, 

I'm exploring the capabilities of Diospyros, specifically focusing on whether it supports logical operations like XOR. I observed that logical XOR is used in the `fft.c` sample file, and it works perfectly in that context. However, I ran into an issue when I tried to implement the `AddRoundKey` kernel in AES.

Here is a simplified snippet of the `AddRoundKey` kernel that causes the problem:
```c
#define SIZE 16

void AddRoundKey(int state_out[SIZE], int state_in[SIZE], int roundKey_in[SIZE]) {
    for (int i = 0; i < SIZE; ++i) {
        state_out[i] = state_in[i] ^ roundKey_in[i];
    }
}

``` 

Steps to Reproduce
1. Save the above code snippet in a file called `aes.c`.
2. Run `cdios ./aes.c`.

The following error message is displayed:
```bash
CDIOS: Compiling C->Racket failed
bitwise-xor: contract violation
  expected: exact-integer?
  given: state_in$0
  context...:
   /home/username/.local/share/racket/8.6/pkgs/rosette/rosette/base/form/control.rkt:15:25
   body of (submod "/home/username/diospyros/src/c-meta.rkt" main)
#f
``` 

I'm wondering if Diospyros currently supports logical XOR operations, or if there's a specific way I should go about using them.

Thank you for your time and assistance.

avanhatt, 2023-09-22:
Hi, and thanks for your interest in the project!

The C frontend for Diospyros is currently only able to use XOR operations on concrete values (as in `fft.c`, where the `^` is between constants or variables concretely defined in each loop iteration). However, it doesn't currently support XOR on integers that may be symbolic (as they are in your example, since `state_in` and `roundKey_in` are arguments to the function). 

Without digging into the code too closely, this is presumably because `bitwise-xor` is _not_ a supposed integer operation in Rosette: https://docs.racket-lang.org/rosette-guide/sec_bools_ints_reals.html.

However, xor on bit vectors _is_ defined in Rosette: https://docs.racket-lang.org/rosette-guide/sec_bitvectors.html#%28def._%28%28lib._rosette%2Fbase%2Fbase..rkt%29._bvxor%29%29

So, if you wanted to support this use case, you could patch Diospyros to convert the integer to a bitvector and call the bitvector version of the function in that case. I'd be happy to review a PR on that, if it's something you are interested in doing.

jingyao-zhang, 2023-09-26:
Thanks for your reply!
Your information and insights are very valuable. I will try to make this `bitwise-xor` and maybe other complex operators work, and definitely PR.

closed 2023-09-26 by jingyao-zhang